terjeio/Trinamic-library #6: Possible bug. Stallguard not functioning without pwm_autoscale

wakass, 2023-01-23:
Stallguard does not function properly (or as expected?) on a 2209 driver. The following line is relevant, and setting autoscale to true recovers proper stallguard behaviour on my system: https://github.com/terjeio/Trinamic-library/blob/9eae92beab071f00d188f9da25672ab9f2aaa116/tmc2209hal.c#L162

I'm not sure how this works in other driver modules, but I can't find any reference on stallguard requiring disabling pwm_autoscaling. Just tcoolthrs < tstep < tpwmthres

troth530, 2023-01-26:
@wakass I tend to agree, in fact it seems the drivers will operate in open loop with velocity based current scaling (requiring motor specific settings) with pwm_autoscale set to false.  In this context the motors will stall during jogging regardless of stallguard threshold.  On the other hand when pwm_autoscale is set to true stalling is only occurring during homing with reasonable stallguard threshold settings that work with my motors.

philletourneau, 2023-01-30:
👀 I've been trying to get stallguard working for days! I'll give this a try too.

Otherwise I'm having trouble with are setting the tCoolThrs and TPWMThrs to any value other than 0, and the driver status is always reporting stealth as 0 (meaning spreadcycle and not stealthchop)

WheeSci, 2025-01-02:
It looks like this boolean is set to true in the version of grblHAL I downloaded in Dec 2024. Thanks!

troth530, 2025-01-02:
> It looks like this boolean is set to true in the version of grblHAL I downloaded in Dec 2024. Thanks!

It was fixed a year ago.

closed 2025-01-04 by terjeio