terjeio/Trinamic-library #14: Cannot reach max current (TMC2209)

paukstelis, 2024-11-03:
In older builds I could set the max run current for TCM2209 ([PLUGIN:Trinamic v0.16]) to 2000 mA:
```
Send: M122
Recv: [TRINAMIC]
Recv:                       X       Z       A       B
Recv: Driver          TMC2209 TMC2209 TMC2209 TMC2209
Recv: Set current        2000    2000    2000    2000
Recv: RMS current        1767    1767    1767    1767
Recv: Peak current       2498    2498    2498    2498
Recv: Run current       31/31   31/31   31/31   31/31
Recv: Hold current      27/31   23/31   27/31   23/31
Recv: CS actual         27/31   23/31   27/31   23/31
Recv: PWM scale            31      27      31      27
Recv: vsense          0=0.325 0=0.325 0=0.325 0=0.325
Recv: stealthChop       false   false   false   false
```
With latest version, I can only set max current up to a certain point. That point is dependent  on the hold current. If I set hold current to 100%, then I can get current up to 1767, at which point I can decrease the hold current again. If I don't set that hold current to 100%, I start to get  out-of-range errors earlier when trying to set run current.

paukstelis, 2024-11-27:
This seems to still exist. I verified that I've pulled the latest.

terjeio, 2024-12-11:
Sorry for the late reply, somehow notifications were turned off for this repo.

> In older builds I could set the max run current for TCM2209 ([PLUGIN:Trinamic v0.16]) to 2000 mA:

Apparently you could but max was still capped by the driver. Current is controlled by CS and vsense, and Rsense limits the max. See chapter 8 in the [datasheet](https://www.analog.com/en/products/tmc2209.html)l. 1767 mA max indicates that you have a 110 milliohm sense resistor. If you have a driver with a lower value resistor this has to be changed compile time

So this is due to adding validation of the current setting in the plugin instead of relying on the driver capping it.

closed 2024-12-12 by terjeio