terjeio/Trinamic-library #15: Why do I take the inverse 2 times?  bool ok = !!tmcdriver[motor.id];

wdfk-prog, 2024-11-21:
https://github.com/terjeio/Trinamic-library/blob/d261eb7393126f0aeb02114c5cc881c7e78728d4/tmc5160hal.c#L372

- Why do I take the inverse 2 times?

wdfk-prog, 2024-11-21:
https://github.com/terjeio/Trinamic-library/blob/d261eb7393126f0aeb02114c5cc881c7e78728d4/tmc5160.c#L151

- TMC5160Reg_ENCM_CTRL No definition found

terjeio, 2024-12-11:
Sorry for the late reply, somehow notifications were turned off for this repo.

> Why do I take the inverse 2 times?

I use it as a shorthand for `bool ok = tmcdriver[motor.id] != NULL;`  See this [post](https://stackoverflow.com/questions/10307281/c-operator-is-a-two-not).

> TMC5160Reg_ENCM_CTRL No definition found

I did not bother to add it since I had no need for it.