terjeio/Trinamic-library #16: information and request

HCY06, 2024-12-12:
i have stm32f103c8t6 mcu, tmc5160 and NEMA 17 stepper motor. may i use that libraries on my project? does it work? the telecommunication is spi between tmc and stm32, and also do you have any examples how to use that libraries on stm32? i just get stuck on project right now. guide me pls thx

HCY06, 2024-12-13:
waiting up

terjeio, 2024-12-13:
You can use the library but you will have to supply your own low level code for communication with the drivers. [This is how I do it](https://github.com/grblHAL/STM32F4xx/blob/master/Src/tmc_spi.c) for the F4xx which is simlar to the F103.

The [tmc5160hal.c](https://github.com/terjeio/Trinamic-library/blob/master/tmc2660hal.c) code was written to provide a higher level API for grblHAL and is used by the [motors plugin](https://github.com/grblHAL/Plugins_motor/tree/531fae8350badbc616d81d7ce973822599ad9c16). If this API is to be used some data structures and possible code from grblHAL has to be extracted. The motors plugin is even tighter integrated into grblHAL, not sure if it would be of any help to study that.