terjeio/Trinamic-library #17: Issue with SPI 5160 initialising

Freezerohmatic, 2025-01-21:
I'm having a strange issue with initialising my 5160 drivers.

I can enable them one by one with some difficulty (by enabling only one axis at time and spamming M122 I a few times)

I change https://github.com/grblHAL/Plugins_motor/blob/87b9056327a87e7a7784a4c9addfb046050199f7/trinamic.c#L2415 to 500 like it was suggested in https://github.com/dresco/STM32H7xx/issues/29. I even change it to 5000 as a sanity check. 

If I remove the check at  https://github.com/terjeio/Trinamic-library/blob/4b86246ef12dcf5d92906e3f077bc433d9e8a8cb/tmc5160.c#L224 changing it to always return 1 if it gets to that point and the stepper driver seems to opperate as expected. 

I feel like I'm missing something as other people seem to not have this issue. 

I'm  using a BTT Octopus Pro v1.1  with STM32H723. All other functions work (execpt for sensorless homing but that seems unrelated)

terjeio, 2025-01-21:
Same issue as [this](https://github.com/grblHAL/Plugins_motor/issues/19)?

Can you try with the latest commit which retries initialization for 5 seconds?
If it still fails what is the output from `$SDS`? And is the `$SDS`? output the same every time reint fails?

terjeio, 2025-01-22:
I have now tested TMC5160 with a STM32F466 board and that works. So the likely reason for failure is in the lowlevel SPI driver code or some hardware issue with your board.

Freezerohmatic, 2025-01-22:
It might be related but no amount of "power on time" with a soft restart fixes my issue. However once initialised it persists after a hard restart but fails again on a power cycle. I've updated (had some issues with GIT and a conflict with the SD card lib). 

$SDS returns error 2   I feel like i'm missing something here

I see your latest message and I have to agree with the level code issue. I have tried playing with timings with no effect. I'm connecting my cro to see whats actaully being outputed. I highly doubt its a wiring issues since once they are "initialised" they work as expected (execpt for homing which is strange since the limit pin is triggered as expected when using iosender Trinamic Tuner tab) and the SPI signals look very clean as well as CS. 

What clock speed are you running the STM32F466? I'm running the STM32H723 at 550MHz.

Freezerohmatic, 2025-01-22:
Figured it out. 

CS pins are changing to fast. Adding a delay after very CS pin change seems to have fixed this

terjeio, 2025-01-22:
> $SDS returns error 2 I feel like i'm missing something here

The $SDS command was added a few days so your code might not have it.

> What clock speed are you running the STM32F466? I'm running the STM32H723 at 550MHz.

Somewhere around 150MHz, the SPI clock at around 3MHz.

Ping to @dresco and @raoullector

Freezerohmatic, 2025-01-22:
150Mhz makes sense for this error. 

Whats strang is that it does not seem like a timing issue for the 5160.... the CS pin must be selected 10ns before any data can be read or written to the driver but what I'm seeing the CS pin not being set if a SPI register is filled to quickly after. Using this delay function https://github.com/dresco/STM32H7xx/blob/82a22312bf06cd76fcd569b9bd5f9ab94f47cdb4/Src/tmc_spi.c#L41 I set it to 15 and some times it works. I leave it at 45 and I've yet to see a failure. When the CS pin is set I see a delay of 4-4.4 us (15 vs 45)

Trying to find a reason for this in the datasheet but can't see anything concrete other the maybe the IO matrix being much slower then the system clock?

@terjeio, should I close this and move to the other thread since this issue is not related to this repo?

terjeio, 2025-01-22:
>  maybe the IO matrix being much slower then the system clock?

IIRC it is.

> should I close this and move to the other thread since this issue is not related to this repo?

Yes.

closed 2025-01-22 by Freezerohmatic

dresco, 2025-01-22:
Hi @Freezerohmatic 

Thanks for your investigations. I've just had a quick look at your repo. Seems you are currently running at 480MHz, same as the rest of the boards (I kept them all the same for consistency), and using hardware SPI for the Trinamic drivers.

Can pick this up in the other issue, but a couple of thoughts off the top of my head;

- It may be instruction cache related, you could try adding a __dsb() call between the CS write and the spi_xxx() call, or just read back the CS pin state.
- Or disable cache completely (for testing) by compiling with L1_CACHE_ENABLE set to 0.
- You could try software SPI instead, just as a comparison (see the map files for some other boards).

One of my dev boards is set up for easy changes between hw & sw SPI, will investigate further on that.

I do have an outstanding [issue](https://github.com/dresco/STM32H7xx/issues/30) to add Octopus Pro 1.1 support in the H7 repo, will take a better look at your changes later..

Cheers,
Jon